LSatan/SmartRC-CC1101-Driver-Lib #100: Somfy open / close button

rkotulan, 2022-01-24:
Hi,
I have additional two buttons for rolling up and down on my Somfy remote. 
(I can control standard buttons.)

With your script https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/44#issuecomment-693636925
I changed frequency to 433.42 MHz

I was able to receive this data:

Counter: 299
10026,15112,2463,2551,2473,2581,2451,2559,2481,2558,2457,2563,2458,2567,2457,2586,2462,2557,2463,2580,2437,2569,2461,2570,2461,2568,4790,1302,1241,1268,1239,706,596,690,610,693,592,1297,1282,692,596,700,606,1290,619,644,1238,689,619,1270,1261,682,621,669,609,1284,620,665,611,659,614,651,618,647,609,660,1255,670,607,680,617,696,601,1280,614,672,620,652,1261,677,623,678,616,675,624,675,616,673,626,1279,1231,698,601,681,628,1279,597,675,1239,691,591,1298,1234,704,595,701,627,1278,622,653,1237,689,592,1295,612,660,623,619,1218,671,587,683,573,691,568,1271,1200,680,576,692,567,696,579,1247,610,611,636,640,1207,690,595,667,585,680,579,691,568,676,577,1266,1209,1266,606,614,1230,4772,2463,2573,2481,2570,2446,2578,2445,2587,2447,2563,2450,2569,4777,1296,1240,1289,1220,701,615,682,593,695,609,1287,1242,698,590,692,616,1295,587,662,1241,701,601,1279,1240,694,609,705,617,1294,585,688,600,674,596,673,585,663,625,666,1220,695,593,711,596,697,582,1318,591,670,610,661,1228,690,605,707,602,688,597,697,584,714,593,1293,1240,690,593,694,609,1291,596,664,1235,711,596,1301,1206,715,595,692,610,1284,616,648,1247,706,575,1302,601,677,611,638,1192,717,540,696,582,695,560,1257,1190,703,548,719,559,703,565,1258,619,616,624,628,1184,706,564,680,572,692,587,674,563,689,594,1246,1223,1258,611,610,1211,5548692,10045,15148,2505,2539,2461,2582,2488,2524,2485,2546,2466,2546,2472,2552,2470,2583,2473,2537,2477,2561,2479,2535,2483,2544,2481,2568,4787,1279,1259,1270,1253,665,623,668,631,1261,629,646,1269,660,628,673,622,1281,634,615,1271,673,611,664,646,669,635,665,625,669,628,1259,640,619,645,628,637,658,1240,1271,1261,678,617,674,636,663,620,1268,637,638,615,654,1246,675,635,665,620,668,

Could you please help me decode it? Or is it necessary to different setup for the receiver?

Thank you,
Rudolf

LSatan, 2022-01-25:
Hi, somfy uses rolling code. I have already successfully decrypted simu (sister company). can you send me the exact name of the remote control? I originally had the plan to integrate somfy into my simu remote. However, I do not have such a remote control. this would be necessary for correct implementation. regards

rkotulan, 2022-01-25:
Telis 4 Modulis RTS5 Channel
https://www.amazon.co.uk/MODULIS-5-channel-control-scroll-wheel-catalogue/dp/B074CB9L47

According to this https://pushstack.wordpress.com/somfy-rts-protocol/ I can control buttons:
- My  - OK - Stop or move to favourite position
- Up - only one step - middle button
- Down - only one step - middle button
- Prog - Used for (de-)registering remotes
The problem is with buttons right up and left down for rolling up and down.