LSatan/SmartRC-CC1101-Driver-Lib #104: New V2.0 E07-M1101D-SMA CC1101

haxti, 2022-03-15:
Hi,

I banged my head on the desk for 2 days now and even ordered more CC1101 modules before I realized, that the new modules seem to have a different pinout.
I ordered this one from Amazon: https://www.amazon.de/Ebyte-E07-M1101D-SMA-Transceiver-Transmission-Empf%C3%A4nger/dp/B07P8S9M4W

They are labeled with V2.0 as seen on the Amazon image.
I just found a alibaba listing and thought I would let you know: https://www.alibaba.com/product-detail/Cleqee-2-E07-M1101D-SMA-CC1101_62358264393.html

The Board looks identical and the pinout in this listing seems to work so far. I only tried the rc-switch DIP example cause this is the one I need.

Thank you for your work. Please let me know if you need more info :)

NorthernMan54, 2022-03-16:
In my collection of CC1101 modules, I have one that uses 8 pins and another that uses 10 pins.  For these I ended up validating the connections between the connector and cc1101 chip as I was not sure what pin was which.  For the 8 pin module I have I did find this data sheet for it that mapped the pins.

https://github.com/NorthernMan54/rtl_433_ESP/blob/master/docs/E07-M1101D-TH_Usermanual_EN_v1.30.pdf

haxti, 2022-03-18:
Yep this matches with my module and the alibaba listing. Thanks for the datasheet.

It's different to the D-Sun Module on LSatan's image https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/img/Wiring_CC1101.png

This issue is just to raise awareness :)

Tomasinjo, 2022-11-15:
Thank god I am not the only one. I just received the M1101D, trying to make it work with NodeMCU v3 (ESP8266).

I initially wanted to control 303MHz device, but I don't think this can be done with M1101D.

For that reason, I wanted to use them to **listen** to some other 433MHz devices (chinese motion sensors). 
I can't get basic example to work - I tried [cc1101_New_Receive_method_minimal](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/CC1101%20default%20examples/New_Method_without_GDO/cc1101_New_Receive_method_minimal), got Connection OK and Rx Mode output, but that is it. 
Everything should be connected correctly - I was referring to attached datasheet and LSatan's scheme for ESP8266.

Before I go on with troubleshooting, am I missing anything obvious?

Tomasinjo, 2022-11-15:
Ok so to respond to myself: Change the following two settings from 1 to 0 and you should see some output:
    ELECHOUSE_cc1101.setSyncMode(0); 
    ELECHOUSE_cc1101.setCrc(0);

Still no luck with 303MHz.. I used protocol analyzer mentioned by LSatan himself (https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/38#issuecomment-665694481), but can't see any output at 303.94MHz.

Can anyone confirm that M1101D doesn't work with 303MHz?

Beastskyler, 2022-11-26:
> Ok so to respond to myself: Change the following two settings from 1 to 0 and you should see some output: ELECHOUSE_cc1101.setSyncMode(0); ELECHOUSE_cc1101.setCrc(0);
> 
> Still no luck with 303MHz.. I used protocol analyzer mentioned by LSatan himself ([#38 (comment)](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/issues/38#issuecomment-665694481)), but can't see any output at 303.94MHz.
> 
> Can anyone confirm that M1101D doesn't work with 303MHz?

Thank you, I have no idea what I am doing. Did you have any luck with 303.94MHz?

Tomasinjo, 2022-11-27:
I haven't tried since then. I kind of lost hope after protocol analyzer didn't output anything.

Maybe OP has more info on that?  @haxti @NorthernMan54

NorthernMan54, 2022-11-28:
I used to have devices at 303 MHz, but have long since replaced them with sonoff ifan03 modules.  What I don’t recall was if I had ever used a cc1101 to control them.

What I can add to this is that at 315mhz I have devices with a cc1101, running openmqttgateway for transmitting the RF codes to control a fireplace.  It will also receive data at 315mhz, but the range is very short ie < 5 meters

Tomasinjo, 2022-11-29:
Thanks for info.
Well I just figured out that my device is not using 303MHz, but 434Mhz. It works well with M1101D.