LSatan/SmartRC-CC1101-Driver-Lib #114: Question: Possible to use multiple Receiver libraries at once?

fightforlife, 2022-12-30:
Die someone try to use multiple Receiver libraries like rc-switch + espilight at once?

I think subscribing to the same pin from two libraries will not work. But maybe I could split the receiver Pin and connect it to two esp32 Input pins in parallel.

 Any ideas on this?

LSatan, 2023-03-01:
yes it is possible. the easiest way is to make a small loop where the reception queries of the individual libraries are inside. possibly there must be a delay between them. therefore it is not the most elegant solution as you run the risk of missing something. the best way is to record the data and then send it to the individual libraries. However, this requires an adjustment in the libraries so that you can receive your recording.

fightforlife, 2023-04-10:
Just as a short feedback here. I wanted to do this for tasmota integration.
I decided to physically split the GDOs to multiple PINs and give each library their own pin they can attach an interrupt to.
My first working prototype is here: https://github.com/fightforlife/Tasmota/tree/development_newSmartRC
With discussion here: https://github.com/arendst/Tasmota/discussions/14343#discussioncomment-5513546

closed 2023-04-10 by fightforlife