LSatan/SmartRC-CC1101-Driver-Lib #118: how to connect to ESP32-C3

Emilevanderlaan, 2023-02-28:
How do I Connect the CC1101 to the ESP32-C3

To setup the CC1101 I cannot use the setSpiPin() as this will be reset by the init()->setSpi().
And there the ESP32 settings are SCK_PIN = 18; MISO_PIN = 19; MOSI_PIN = 23; SS_PIN = 5; will be set.
But that is not the setup for the ESP32C3 or ESP32S2
support for the gpio_iomux_out / gpio_iomux_in is also missing.
Please advice how to connect the CC1101

<html>
<body>
<h1 tabindex="-1" dir="auto" style="box-sizing: border-box; font-size: 2em; margin: 24px 0px 16px; font-weight: var(--base-text-weight-semibold, 600); line-height: 1.25; padding-bottom: 0.3em; border-bottom: 1px solid var(--color-border-muted); color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Wirering</h1>

CC1101 |   | ESP32 | ESP32-S2/S3 | ESP32-C2/C3
-- | -- | -- | -- | --
MISO | -- | GPIO19 | GPIO37 | GPIO4
SCK | -- | GPIO18 | GPIO36 | GPIO3
MOSI | -- | GPIO23 | GPIO35 | GPIO2
CSN | -- | GPIO5 | GPIO34 | GPIO1
GDO0 | -- | GPIO15 | GPIO33 | GPIO0
GDO2 | -- | N/C | N/C | N/C
GND | -- | GND | GND | GND
VCC | -- | 3.3V | 3.3V | 3.3V

<br class="Apple-interchange-newline">
</body>
</html>

feyhong1112, 2024-03-29:
Hi do know which pin is used to connect gdo2 currently I use pin 3 instead 4 but It doesnt work

Emilevanderlaan, 2024-04-01:
Feyhong Send me a Email (Fey [At] emsign [dot] nl) then i will send you what i have done, it is not fit to put on Github