LSatan/SmartRC-CC1101-Driver-Lib #120: How to use on Arduino UNO and how it works?

Drunk-Lamp, 2023-03-15:
Hello, i am a noob at arduino and i just bought the cc1101 but ith a new version or some beacuse i have 8 pins in total and i dont know how to wire it to a arduino uno. Can some one mayby also explain the basics of this library beacuse i always tought that you need to set the pins in the code but in the example codes not any pins are given.

NorthernMan54, 2023-03-16:
This may help, I found this datasheet for an 8 Pin CC1101 module I had purchased

https://github.com/NorthernMan54/rtl_433_ESP/blob/master/docs/E07-M1101D-TH_Usermanual_EN_v1.30.pdf

Drunk-Lamp, 2023-03-17:
but how do i connect this to a arduino uno, beacuse in the examples no pins are definend so i dont know how to wire it

Gemikro1, 2023-03-18:
I would maybe just scroll down a little bit on the main repository page to the section called wiring ...