LSatan/SmartRC-CC1101-Driver-Lib #124: Error: Guru Meditation Error: Core  1

frspp, 2023-04-06:
When pressing certain button on my four button remote I get
```
CC1101: Connection OK
CC1101: Rx mode
Rssi: -65
LQI: 255
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40085ef2  PS      : 0x00060930  A0      : 0x800d2c44  A1      : 0x3ffb1ec0  
A2      : 0xffff8cff  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000006  A8      : 0x80085c30  A9      : 0x3ffb1ea0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffff8d3f  LBEG    : 0x400d2bec  LEND    : 0x400d2bf9  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40085ef2:0x3ffb1ec0 0x400d2c41:0x3ffb1f00 0x400d1d14:0x3ffb1f20 0x400d109d:0x3ffb1f40 0x400d1b5a:0x3ffb1f60 0x400d0e17:0x3ffb1f90 0x400d387d:0x3ffb1fb0 0x400861c1:0x3ffb1fd0

Rebooting...
```

jayateertha043, 2023-06-18:
Try increasing buffer length to 300 it may work, for some reason 61 buffer length gives this error

hipotenusa, 2025-01-21:
What actually solved this error in my case was ensuring spi pins were correctly set by calling `ELECHOUSE_cc1101.setSpiPin();` before `ELECHOUSE_cc1101.getCC1101();`