LSatan/SmartRC-CC1101-Driver-Lib #127: Finally got it working - setting custom pins - NodeMCU ESP32

frspp, 2023-04-07:
I have NodeMCU ESP32. Using default pins introduced flash MD5 hash error so I chose custom pins. I recommend using **GPIO pins 16-33** because pins 0-15 have some special functions and caused problems (see https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ )

Provided examples were not working and documentation is not solid. Took long to find these needed modifications to make it work: 
- setSpiPin() before init()
- setGDO() before init()
- if-else getCC1101() AFTER the settings, not before like in example.

Working Arduino sketch: 

```
/*
  Example for receiving
  
  https://github.com/sui77/rc-switch/
  https://github.com/LSatan/SmartRC-CC1101-Driver-Lib
  If you want to visualize a telegram copy the raw data and 
  paste it into http://test.sui.li/oszi/
  ----------------------------------------------------------
  Mod by Little Satan. Have Fun!
  ----------------------------------------------------------
*/
#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include <RCSwitch.h>

int pin; // int for Receive pin.

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);

pin = 33;  // GDO2 (33 for me)

  ELECHOUSE_cc1101.setSpiPin(18, 19, 23, 17); // (SCK, MISO, MOSI, CSN). Must be BEFORE Init()!
  ELECHOUSE_cc1101.setGDO(32, 33); //Set Gdo0 (tx) and Gdo2 (rx) for serial transmission function.
  ELECHOUSE_cc1101.Init();            // must be set to initialize the cc1101!
  //ELECHOUSE_cc1101.setRxBW(812.50);  // Set the Receive Bandwidth in kHz. Value from 58.03 to 812.50. Default is 812.50 kHz.
  //ELECHOUSE_cc1101.setPA(10);       // set TxPower. The following settings are possible depending on the frequency band.  (-30  -20  -15  -10  -6    0    5    7    10   11   12)   Default is max!
  ELECHOUSE_cc1101.setMHZ(433.92); // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
  
  mySwitch.enableReceive(pin);  // Receiver on interrupt 0

  ELECHOUSE_cc1101.SetRx();  // set Receive on

  if (ELECHOUSE_cc1101.getCC1101()) {       // Check the CC1101 Spi connection.
    Serial.println("Connection OK");
  }
  else {
    Serial.println("Connection Error");
  }
}

void loop() {
  if (mySwitch.available()) {
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}

```

gc9n, 2023-11-02:
it doenst work either !!!!

BeniIgle, 2023-11-26:
It says connection error

BeniIgle, 2023-11-29:
I dont receive any code either

BeniIgle, 2023-11-30:
Could you please explain how do i connect my cc1101 to my esp32? i mean like putting 1- gnd 2- 3.3v
thanks and sorry, im really new in this world