LSatan/SmartRC-CC1101-Driver-Lib #138: Esp-Wroom-32 CC1101 (8pins)

BeniIgle, 2023-12-04:
Hello there, could someone please help me to connect and give me a code example?
i have my esp-wroom-32 and my 8pin cc1101 (EO7-M11010) and i want it to receive 433.92 mhz signals
i tried every single connection and code in the internet and no one worked.
Im thinking that the problem could be the power(? 
i dont know what else to think or to do.
even tho sometimes it puts connection ok, i receive random data every second and when i press my remote buttton, i dont receive nothing but the random codes

[image]

CarLoOSX, 2024-01-10:
check your data sheet should be something like this one : https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjE7_TwyNODAxU8Q6QEHcx3ACAQFnoECBcQAQ&url=https%3A%2F%2Fwww.ebyte.com%2Fen%2Fdownpdf.aspx%3Fid%3D165&usg=AOvVaw0_nU7dTRDlzfCF9m1j1tQn&opi=89978449

then you need to setup the pins with this functions, before init:

ELECHOUSE_cc1101.setSpiPin(18, 19, 23, 5);
ELECHOUSE_cc1101.setGDO(2, 4);

smolbun, 2024-03-15:
@slater0013 May I know what are you trying to accomplish? I manage to get the module to replay signals captured from an RF remote, is this what you're attempting as well?

slater0013, 2024-03-15:
Just trying to build a long range powerpoint remote here :) previous/next buttons

smolbun, 2024-03-15:
Are you building one from scratch? Or is it an existing powerpoint remote that you're using?

slater0013, 2024-03-18:
Building one from scratch ! I was looking for a project idea to start digging into the wireless modules :)

smolbun, 2024-03-18:
You could take a look at the [example for tx and rx](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master/examples/CC1101%20default%20examples/New_Method_without_GDO) and set the pins accordingly for your ESP32.

` ELECHOUSE_cc1101.setSpiPin(SCK_PIN, MISO_PIN, MOSI_PIN, SS_PIN);` this should be called before `ELECHOUSE_cc1101.Init();` As for `ELECHOUSE_cc1101.getCC1101()`, it seems to always return true for some reason.

CarlosDanielDev, 2024-03-29:
Hello there!!

When attempting to run the minimal receiver example from the Arduino library for the CC1101 module, I encountered several compilation errors and warnings. These occurred in the ELECHOUSE_CC1101_SRC_DRV.cpp file.

## Steps to Reproduce:

- Opened the minimal receiver example in the Arduino IDE.
- Compiled the code for my esp32.

## Error Messages:
The specific errors and warnings I encountered are:

```bash
ELECHOUSE_CC1101_SRC_DRV.cpp:149:3: error: this 'while' clause does not guard... [-Werror=misleading-indentation]
   while(digitalRead(MISO_PIN));
   ^~~~~
   
   
ELECHOUSE_CC1101_SRC_DRV.cpp:150:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'while'
  digitalWrite(SS_PIN, HIGH);
    ^~~~~~~~~~~~
  
  ELECHOUSE_CC1101_SRC_DRV.cpp:533:11: warning: comparison is always false due to limited range of data type [-Wtype-limits]
 if (freq0 > 255){freq1+=1;freq0-=256;}
     ~~~~~~^~~~~
     
     ELECHOUSE_CC1101_SRC_DRV.cpp:983:5: error: unused variable 's2' [-Werror=unused-variable]
 int s2 = 0;
     ^~
     
     
     ELECHOUSE_CC1101_SRC_DRV.cpp:500:13: error: 'a' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 PA_TABLE[0] = a;
 ~~~~~~~~~~~~^~~
cc1plus: some warnings being treated as errors
```

I would appreciate any help in understanding and resolving these issues.

closed 2026-04-26 by github-actions[bot]