LSatan/SmartRC-CC1101-Driver-Lib #152: The example sketches no longer contain the frequency calibration (v2.5.7)

Pierre-Serrier, 2024-04-03:
Hello,
Where can I find the example sketch Arduino to calibrate (in freqency) my cc1101 ?
Thanks for your answer.
Pierre

feyhong1112, 2024-04-08:
I am not sure but in new lib I see it have the section include calibrate the frequency. Hope this help

feyhong1112, 2024-04-08:
Here:
[image]
This is the function that connect to code:
[image]

Pierre-Serrier, 2024-04-10:
Thanks feyhong1112 but what I'm looking for is the sketch that allows you to calibrate each CC1101. That is to say to position the values ​​of clb1[x], clb2[x] , clb3[x] and clb4[x] (with x=0 or 1). A previous version of the library provided this sketch in the examples.

Pierre-Serrier, 2024-04-18:
I found him.  My apologies for the inconvenience.
[image]