LSatan/SmartRC-CC1101-Driver-Lib #160: Hang in Init

chrisonair, 2024-07-05:
I get the same thing. Tried 4 different cables, different USB ports. I cant get passed the initialization.. spi says connection ok. Then i do the init and it hangs there

I edited elechouse ccp and in init i added some lines to debug:
`  Serial.println("Connection OK1");
  setSpi();
  SpiStart();                   //spi initialization
  Serial.println("Connection OK2");
  digitalWrite(SS_PIN, HIGH);
  Serial.println("Connection OK3");
  digitalWrite(SCK_PIN, HIGH);
  Serial.println("Connection OK4");
  digitalWrite(MOSI_PIN, LOW);
  Serial.println("Connection OK5");
  Reset();                    //CC1101 reset
  Serial.println("Connection OK6");
  RegConfigSettings();            //CC1101 register config
  Serial.println("Connection OK7");
  SpiEnd();
  Serial.println("Connection OK8");`

I get up to connection OK6. wont get 7 or 8.. so it hands on RegConfigSettings();

Lesateq, 2024-07-08:
I have compiled the library for three board:

1) Wemos 8266 D1 Mini -> Works well, although I need to explicitely use the setSpiPin(byte sck, byte miso, byte mosi, byte ss) function. I have used the default SPI pins, but the code only works when using SetSpiPin()

2) NodeMcu ESP32 Wroom Wifi/Bluetooth-> Works well using default SPI pins

3) ESP32C3 Super Mini -> this is a nigtmare. I work with the standard SPI pins, and use the SetSpiPin() to make sure the pins are configured right. Next the code does not run and enters an 'infinite loop' when executing the init() routine. And indeed, I found out that it crashes on the SpiWriteReg() and other Spi routines. I verified that the board is okay. Another CC1101 library (https://github.com/wladimir-computin/CC1101-ESP-Arduino) does actually work. If anybody has a clue why the ESP32C3 Super Mini does not work please reply.

chrisonair, 2024-07-08:
Hi. Esp32s3 did compile but couldnt pass the init. I had to change a bit
your code to work.

Thanks!

closed 2026-04-26 by github-actions[bot]