LSatan/SmartRC-CC1101-Driver-Lib #163: Error: "__digitalRead(): IO 19 is not set as GPIO.", even with example code

PKeidel, 2024-08-13:
Hi,
I am trying to use a C1101 to receive values from my water meters.

While testing with the most simple example "c1101_New_Receive_method_minimal" and the default pins, I got the following esp32 error:

`[    13][E][esp32-hal-gpio.c:190] __digitalRead(): IO 19 is not set as GPIO.`

Is this problem already known?

I added some logging to "SmartRC-CC1101-Driver-Lib/ELECHOUSE_CC1101_SRC_DRV.cpp" and the `pinMode()` calls in `ELECHOUSE_CC1101::SpiStart` do look correct.

```c++
  Serial.printf("SpiStart() outputs= %d, %d, %d; inputs:%d\n", SCK_PIN, MOSI_PIN, SS_PIN, MISO_PIN);
```
Prints: `SpiStart() outputs= 18, 23, 5; inputs:19`

```
FQBN: esp32:esp32:esp32doit-devkit-v1:DebugLevel=info,EraseFlash=all
Using board 'esp32doit-devkit-v1' from platform in folder: ...
Using core 'esp32' from platform in folder: ...

Using library SmartRC-CC1101-Driver-Lib at version 2.5.7 in folder: /home/.../Arduino/libraries/SmartRC-CC1101-Driver-Lib 
Using library SPI at version 3.0.4 in folder: /home/.../.arduino15/packages/esp32/hardware/esp32/3.0.4/libraries/SPI 

esptool.py v4.6
Chip is ESP32-D0WDQ6 (revision v0.0)
```

panaak, 2024-09-28:
I'm having a similar issue, have you figured something out?

PKeidel, 2024-10-09:
I had no success. But a different error. Now I get:
```
E (1637) gpio: gpio_set_level(227): GPIO output gpio_num error
[  1652][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
```

dewenni, 2024-12-01:
@panaak @PKeidel @LSatan 
any news on this topic?
I have the same problem in combination with Arduino V3.0.7 and ESP32

for all SPI PINs I get errors like this:
`[esp32-hal-gpio.c:190] __digitalRead(): IO 19 is not set as GPIO.`

SinekaRamu, 2025-01-07:
@dewenni, I have the same issue in combination with Arduino V2.3.4 and ESP32-cam

For all SD card pins I got this same error:
`[esp32-hal-gpio.c:190] __digitalRead(): IO 15 is not set as GPIO.`

did you found any solution?

panaak, 2025-01-07:
Well, for me it never was a real problem, the error started to show when i tried to set the core debug level.
(tools -> core debug level)
When set to anything but None, the error started to show, but the board basically worked as expected. Turned off the debug level and all was good.
Now the problem is even more distant since i dont use arduino ide anymore. Its general flawed functionality, stability and compile times performance makes it unusable for me.

Viproz, 2025-08-09:
This is due to the use of the SPI, on ESP32 when the SPI library is used the pins are no longer considered as GPIO pins. You have to use a lower level library to do it.

You can see my fix here https://github.com/Viproz/SmartRC-CC1101-Driver-Lib/commit/4eec10ae0dbf38bc31b5b1f8d8119f42cb9b05d6

No clue why it worked before.

closed 2026-04-26 by github-actions[bot]