LSatan/SmartRC-CC1101-Driver-Lib #167: ESP32 WROOM 32 + M1101D module keeps rebooting or hangs almost with all example files

miroslavkrajcir, 2024-11-27:
Hello, i have a ESP32 WROOM 32 board with M1101D (CC1101 Chip) module. When using example files my board keeps rebooting or hangs at setup, i even tried the example files without the C1101 module connected but the problem persists. Any ideas?

ScorpionXII, 2024-12-05:
I was experiencing the same issue. It seems the pins are not correctly configured. Just do this:
`#define SCK_PIN 18
#define SS_PIN 5
#define MISO_PIN 19
#define MOSI_PIN 23`
 
and add this line as first intruction after `void setup() {`
`ELECHOUSE_cc1101.setSpiPin(SCK_PIN, MISO_PIN, MOSI_PIN, SS_PIN);`

nick888888888, 2025-04-28:
I kept getting the boot loop, even with pins configured as per above. Commenting out this portion seemed to fix it. However I still have other problems (e.g. transmit doesn't transmit reliably). Maybe not calling getCC1101() actually breaks all functionality, but I won't have time to look until later.

```
    // if (ELECHOUSE_cc1101.getCC1101()){      // Check the CC1101 Spi connection.
    // Serial.println("Connection OK");
    // }else{
    // Serial.println("Connection Error");
    // }
```

BatrakovSV, 2025-05-02:
> I kept getting the boot loop, even with pins configured as per above. Commenting out this portion seemed to fix it. However I still have other problems (e.g. transmit doesn't transmit reliably). [...]

// Define SPI pins
#define SCK_PIN 18
#define SS_PIN 5
#define MISO_PIN 19
#define MOSI_PIN 23

void setup() {
    // Set SPI pins for CC1101
    ELECHOUSE_cc1101.setSpiPin(SCK_PIN, MISO_PIN, MOSI_PIN, SS_PIN);

    Serial.begin(9600);