LSatan/SmartRC-CC1101-Driver-Lib #170: esp32-c3-supermini  - cc1101 v2  hanging on init

luupux, 2025-04-05:
Hello,  I'm checking to switch from esp32-wroom to esp32-c3-supermini  but I went in stuck, I tried different approach then I was not able to understand what I'm doing wrong and I decide to open an issue for help 
Mainly  the code is in stuck in the init phase,
I already reviewed multiple time the code, wiring and tested  the code in two different board an cc1101 so I don't know
what else do 

#include <ELECHOUSE_CC1101_SRC_DRV.h>
#include "driver/gpio.h"
#include <SPI.h>

#define PIN_SCK   3
#define PIN_MISO  5
#define PIN_MOSI  6
#define PIN_CS    10

void setup() {
  Serial.begin(115200);
  delay(500);

  Serial.println("🔧 Starting transmitter (v2.5.7)...");
  Serial.println("Test SPI");
  SPI.begin();
  Serial.println("ELECHOUSE_cc1101.setSpiPin");
  ELECHOUSE_cc1101.setSpiPin(PIN_SCK, PIN_MISO, PIN_MOSI, PIN_CS);
  Serial.println("Test Init");
  ELECHOUSE_cc1101.Init();  
  Serial.println("test secCCMode");
  ELECHOUSE_cc1101.setCCMode(1);       // set config for internal transmission mode.
  ELECHOUSE_cc1101.setModulation(0);  // set modulation mode. 0 = 2-FSK, 1 = GFSK, 2 = ASK/OOK, 3 = 4-FSK, 4 = MSK.
  ELECHOUSE_cc1101.setMHZ(433.92);   // Here you can set your basic frequency. The lib calculates the frequency automatically (default = 433.92).The cc1101 can: 300-348 MHZ, 387-464MHZ and 779-928MHZ. Read More info from datasheet.
  ELECHOUSE_cc1101.setSyncMode(2);  // Combined sync-word qualifier mode. 0 = No preamble/sync. 1 = 16 sync word bits detected. 2 = 16/16 sync word bits detected. 3 = 30/32 sync word bits detected. 4 = No preamble/sync, carrier-sense above threshold. 5 = 15/16 + carrier-sense above threshold. 6 = 16/16 + carrier-sense above threshold. 7 = 30/32 + carrier-sense above threshold.
  ELECHOUSE_cc1101.setCrc(1);      // 1 = CRC calculation in TX and CRC check in RX enabled. 0 = CRC disabled for TX and RX.
  Serial.println("Rx Mode");
}

void loop(){
  Serial.println("amicici");
}

Then this is the output 
12:14:38.705 -> 🔧 Starting transmitter (v2.5.7)...
12:14:38.705 -> 
12:14:38.705 -> Test SPI
12:14:38.705 -> 
12:14:38.705 -> ELECHOUSE_cc1101.setSpiPin
12:14:38.705 -> 
12:15:26.217 -> 🔧 Starting transmitter (v2.5.7)...
12:15:26.217 -> Test SPI
12:15:26.217 -> ELECHOUSE_cc1101.setSpiPin
12:15:26.217 -> Test Init

MaxiKn, 2025-06-10:
Hey I have the same problem, have found a solution yet ??

Viproz, 2025-09-25:
If you still have some issues you can try my fork https://github.com/Viproz/SmartRC-CC1101-Driver-Lib 
I added some timeouts to some while loops which could cause crashes if the CC1101 does not behave exactly as expected. I also merged the fix for the RegConfig issue.